MHSeals/mhsboat_ctrl #45: PLAN

bryanchan407, 2024-12-09:
LIDAR will be used to identify clusters of potential buoys and their positions, and the LIDAR odometry will calculate the position of the boat. LIDAR Odometry will be useful when the boat cannot see smaller objects (i.e small ball buoys), so that we can backtrack to where we were and translate its position now from what it was when we moved.

Camera will be used to identify the type of buoy the cluster actually is, and will correspond to a python class that Alec created. 

LIDAR odometry can be used, or the integration of the acceleration from the CubeOrange IMU.

Our main method of navigating through buoys, as discussed by me and Alec, would be to set a midpoint between two pairs of buoys for challenge 1, and using the laser in the direct path forward of the boat to guide ourselves to that midpoint. For example, if the midpoint was to the right of the laser, the boat would use the left thruster to guide itself towards that point. Vice versa. 

My idea for challenge 2 was to do the same thing, except that you would calculate the midpoint with the furthest gap (finding the magnitude of the gaps between the yellow buoys and the corresponding red and green buoys, and finding the biggest gap and proceeding towards the midpoint set between the gap.) If you have another idea for this, please reply.

We are no longer planning to use PWM signals to guide the boat, as the CubeOrange already has a /cmd_vel topic to control the boat. Now, I'm not sure what unit the number sent to the topic corresponds to, but that needs to be figured out.

We need to get people on creating scripts.

I want a group of people that will consolidate sensor data for our scripts to run. This will include data from the LIDAR from /velodyne_points, camera. The camera and the lidar should be configured correctly according to Alec, but I'm wary about mavros, because people have been telling me scary things about documentation. **The goal of this is to have every stream of data as a topic communicating on the odroid and readily accessible to receive data from our scripts.**

We need another group of people, using that data from the topics, to create code. Using other libraries and the data we have, implement a solution (or me and Alec's solutions for that matter.) Clone the mhsboat_ctrl repository, and ask questions. Code as if you already had all the data. 

This is the final plan, no more talking about Nav2 or URDF or whatever. We are using the cubeorange, no ardupilot, camera and lidar to take us to the finish line. We can add less important stuff later.

LETS GOOOOOOOO MARTIN SEALS!!! LETS GOOOOOOOO MARTIN SEALS!!

bryanchan407, 2024-12-09:
For sensors, I would like Bennett, Ryan, and Alec to configure those and make sure we can access the data from those topics.

alec-jensen, 2024-12-10:
The cube orange is ardupilot

bryanchan407, 2024-12-11:
So we are still using ardupilot... but not to set waypoints?

closed 2025-05-31 by alec-jensen