MHSeals/mhsboat_ctrl #51: Divide the run function under taskone.py into separate functions to be made by different people

bryanchan407, 2024-12-14:


1unarzDev, 2024-12-18:
To my knowledge, the current structure utilizes init, search, run, and main functions, so the parent script can interpret and use each task. We need to focus on getting the logic for the search and run functions working. The following are functions that we should consider implementing.
- [ ] Sensor data collection (these might just be directly fed as data into the parent function)
- [ ] Pattern recognition (We should seriously separate this into a separate script, but the general logic is outlined below)
    - [ ] Find buoys of a certain type and color
    - [ ] Determine if the clusters are in a certain arrangement
    - [ ] If not currently working on a different task, prepare to start the run function
- [ ] Helper functions (examples below)
    - [ ] Calculate angles between buoys
    - [ ] Interpolate between two locations (this might just be a natural part of using the motors to reach a goal)
    - [ ] Find buoy pairs
- [ ] Task run logic
    - [ ] Change orientation to face midpoint
    - [ ] Send command to start moving or to stop
    - [ ] Match buoy pairs to LiDAR point clusters

bryanchan407, 2024-12-24:
if you need a helper function made, make a new issue with the function specfied if it's not present in the code

otherwise, I agree with the task run logic, but could you elaborate on what match buoy pairs to lidar point clusters is? @1unarzDev